shivasiddharth/Assistants-Pi #165: audio-test.sh problem

Qhair-LUO, 2021-04-29:
Hi @shivasiddharth !
In step.9,  my playback device(HDMI) is work, but when testing the microphones, I get the below error:

`Select your audio config that you used:
1) AIY-HAT-or-CUSTOM-HAT
2) USB-DAC-or-USB-SOUNDCARD-or-USB-MIC

Please select an option [1-2] 2
Selection: USB-DAC-or-USB-SOUNDCARD-or-USB-MIC
=============Recording Mic Audio Sample=============
arecord: main:828: audio open error: No such file or directory

Finished recording the samples.

Playing back the recorded audio sample......

/home/pi/mic-test.wav: No such file or directory
Did you hear the recorded audio sample?
[y/n] >> 
`

shivasiddharth, 2021-04-29:
Skip the audio test. Its just a bonus.

closed 2021-04-29 by shivasiddharth

rajahslimen, 2021-05-07:
hi 
did you find a solution for this erreur(828)?

rajahslimen, 2021-05-07:
i did finish all the steps and the mic didn't work

rajahslimen, 2021-05-07:
when i tested this commande line '/home/pi/env/bin/python -u /home/pi/Assistants-Pi/Google-Assistant/src/main.py --project_id 'replace this with the project id '--device_model_id 'replace this with the model id'
the google assistant works but the mic didn't work

shivasiddharth, 2021-05-07:
USB Mic are plug and play. You need to select the device from the audio icon on the bar above.  Install pavucontrol, open that move over to the output tab and see if your Mic's VU meter is fluctuating. Also check the part about running pulseaudio systemwide.