shivasiddharth/Assistants-Pi #166: arecord: main:828: audio open error: No such file or directory

rajahslimen, 2021-05-07:
Hi
I have a problem with the mic configuration , i'm using a USB mic and a JACK speaker.
In the instalation i did chose 'USB-MIC-JACK'
when i run  "sudo /home/pi/Assistants-Pi/scripts/audio-test.sh  "

Great!! Proceeding to test the microphones......

A 10 second audio sample will be recorded for testing. Are you ready?
[y/n] >> y
Select your audio config that you used:
1) AIY-HAT-or-CUSTOM-HAT
2) USB-DAC-or-USB-SOUNDCARD-or-USB-MIC

Please select an option [1-2] 2
Selection: USB-DAC-or-USB-SOUNDCARD-or-USB-MIC
=============Recording Mic Audio Sample=============
arecord: main:828: audio open error: No such file or directory

Finished recording the samples.

Playing back the recorded audio sample......

/home/pi/mic-test.wav: No such file or directory
Did you hear the recorded audio sample?
[y/n] >> 
How can i fixe it?

shivasiddharth, 2021-05-07:
Audio test is just a bonus script. I have clearly mentioned in the README to not panic if it not works.

closed 2021-05-07 by shivasiddharth

rajahslimen, 2021-05-07:
I did finish all the steps and when i tested the this '/home/pi/env/bin/python -u /home/pi/Assistants-Pi/Google-Assistant/src/main.py --project_id 'replace this with the project id '--device_model_id 'replace this with the model id'' 
the google assistant work but the mic didn't work