shivasiddharth/Assistants-Pi #167: Google Assistant unrecognized

mikecheng2626, 2021-05-08:
Alex installation is perfect ! Thank you

Google installation:
1. Stuck at the final step: the manual startup of google assistant
2. main.py: error: unrecognized arguments: model ID and project ID
3. I copied from the modelID text file and input them into ReadMe file. Copy and pasted the manual command in terminal and the terminal said they're unrecognized

mikecheng2626, 2021-05-08:
Ok! Here you go!

[image]

shivasiddharth, 2021-05-08:
There is no space between the end of your project id and the start of model id

shivasiddharth, 2021-05-08:
Please carefully go through my suggestion. I said that you need to add a space between the end of project id and start of model id. Instead, you have removed the space after --device_model_id.

mikecheng2626, 2021-05-08:
The first picture. There was a space between project id and model id and I thought you meant remove the space, so I removed it and showed it in 2nd picture. With a space between project id and model id, I get the same result just like the 1st time

Below is what I copied directly from your Assistant-Pi github:

/home/pi/env/bin/python -u /home/pi/Assistants-Pi/Google-Assistant/src/main.py --project_id 'replace this with the project id '--device_model_id 'replace this with the model id'

It looks like there's no space between project id and model id

shivasiddharth, 2021-05-08:
As far as I can see, there is no space between the project id and the model id in both your pics. The prompts from the assistant also tells the same thing. If you see the README you can clearly see the space. If you still find it difficult to format the command, type your model id and project id here I will format and attach.

mikecheng2626, 2021-05-08:
Ok. It didn't work when I added a space in terminal. I'm not sure why, but I added a space in README and copy/paste into terminal and it worked.

But, I didn't hear any intro music and Google is not responsive. Alexa is doing well.

shivasiddharth, 2021-05-08:
Check the README about setting pulseaudio systemwide

wgtneth, 2021-05-19:
hi shivasiddharth
I do not have a config.json file. Can you give me this config.json file?