shivasiddharth/Assistants-Pi #168: I installed the google assistant on rasp 3 and it was working fine for the first 2 days..but then it stopped botting up by itself...I tried to run it manually buut it shows this...

biswasdebanga, 2021-07-06:


biswasdebanga, 2021-07-06:
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 29, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 15, in <module>
    from youtube_search_engine import google_cloud_api_key
  File "/home/pi/GassistPi/src/youtube_search_engine.py", line 13, in <module>
    configuration = yaml.load(conf, Loader=yaml.FullLoader)
  File "/home/pi/env/lib/python3.7/site-packages/yaml/__init__.py", line 114, in load
    return loader.get_single_data()
  File "/home/pi/env/lib/python3.7/site-packages/yaml/constructor.py", line 49, in get_single_data
    node = self.get_single_node()
  File "/home/pi/env/lib/python3.7/site-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
  File "/home/pi/env/lib/python3.7/site-packages/yaml/composer.py", line 55, in compose_document
    node = self.compose_node(None, None)
  File "/home/pi/env/lib/python3.7/site-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/home/pi/env/lib/python3.7/site-packages/yaml/composer.py", line 127, in compose_mapping_node
    while not self.check_event(MappingEndEvent):
  File "/home/pi/env/lib/python3.7/site-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  File "/home/pi/env/lib/python3.7/site-packages/yaml/parser.py", line 428, in parse_block_mapping_key
    if self.check_token(KeyToken):
  File "/home/pi/env/lib/python3.7/site-packages/yaml/scanner.py", line 115, in check_token
    while self.need_more_tokens():
  File "/home/pi/env/lib/python3.7/site-packages/yaml/scanner.py", line 152, in need_more_tokens
    self.stale_possible_simple_keys()
  File "/home/pi/env/lib/python3.7/site-packages/yaml/scanner.py", line 292, in stale_possible_simple_keys
    "could not find expected ':'", self.get_mark())
yaml.scanner.ScannerError: while scanning a simple key
  in "/home/pi/GassistPi/src/config.yaml", line 112, column 1
could not find expected ':'
  in "/home/pi/GassistPi/src/config.yaml", line 135, column 16

shivasiddharth, 2021-07-06:
Looks like you have messed with your config file.

biswasdebanga, 2021-07-06:
I never did anything...It just stopped working sometime later.This is not the first time though.This has happened twice before

shivasiddharth, 2021-07-06:
Attach your config.yaml file here.