shivasiddharth/Assistants-Pi #169: Assistants not starting on boot and google assisstant not responding for wakeword

Prasanna-16, 2021-07-11:
Hi sir I followed all your instructions. I did not got a single error in installation. But when I reboot the Rasberry pi 3b both assistants do not started. When I manually started the alexa it works well. But google assistant starts and says hi I manually started but not responds to wakeword.

Prasanna-16, 2021-07-11:
I hadd also tried setup PulseAudio as a system wide service.

shivasiddharth, 2021-07-11:
1. Check the status of the assistant services on boot.
2. Check the status of pulseaudio system wide service on boot.

Syntax for checking the sttaus of service is:
`sudo systemctl status NAME_OF_THE_SERVICE`

Prasanna-16, 2021-07-11:
Sir what should I replace in place of NAME_OF_THE_SERVICE?

shivasiddharth, 2021-07-11:
For alexa it will be:
alexa.service
For Google Assistant it will be:
google-assistant.service 
For pulse audio system wide it will be:
pulseaudio.service

All the details are in the README document.

Prasanna-16, 2021-07-11:
Sir I did this first for alexa it shows like this

>  alexa.service - Amazon Alexa Voice Service
   Loaded: loaded (/lib/systemd/system/alexa.service; enabled; vendor preset: en
   Active: active (running) since Sun 2021-07-11 19:00:10 IST; 2min 14s ago
 Main PID: 537 (screen)
    Tasks: 41 (limit: 2062)
   CGroup: /system.slice/alexa.service
           ├─537 SCREEN -dm sudo python3 /home/pi/Assistants-Pi/Alexa/alexa.py
           ├─539 sudo python3 /home/pi/Assistants-Pi/Alexa/alexa.py
           ├─540 python3 /home/pi/Assistants-Pi/Alexa/alexa.py
           ├─721 sudo /home/pi/Assistants-Pi/Alexa/startsample.sh
           ├─727 sh /home/pi/Assistants-Pi/Alexa/startsample.sh
           └─729 ./SampleApp /home/pi/Assistants-Pi/Alexa/build/Integration/Alex

Jul 11 19:00:09 raspberrypi sudo[478]: pam_unix(sudo:session): session opened fo
Jul 11 19:00:10 raspberrypi sudo[533]:     root : TTY=unknown ; PWD=/home/pi/Ass
Jul 11 19:00:10 raspberrypi sudo[533]: pam_unix(sudo:session): session opened fo
Jul 11 19:00:10 raspberrypi sudo[533]: pam_unix(sudo:session): session closed fo
Jul 11 19:00:10 raspberrypi sudo[478]: pam_unix(sudo:session): session closed fo
Jul 11 19:00:10 raspberrypi sudo[539]:     root : TTY=pts/1 ; PWD=/home/pi/Assis
Jul 11 19:00:10 raspberrypi sudo[539]: pam_unix(sudo:session): session opened fo
Jul 11 19:00:10 raspberrypi systemd[1]: Started Amazon Alexa Voice Service.
Jul 11 19:00:14 raspberrypi sudo[721]:     root : TTY=pts/1 ; PWD=/home/pi/Assis
Jul 11 19:00:14 raspberrypi sudo[721]: pam_unix(sudo:session): session opened fo

And for Google assistant,

> pi@raspberrypi:~ $ sudo systemctl status google-assistant.service
● google-assistant.service - Google Assistant Voice Service
   Loaded: loaded (/lib/systemd/system/google-assistant.service; enabled; vendor
   Active: active (running) since Sun 2021-07-11 19:00:08 IST; 5min ago
 Main PID: 479 (python)
    Tasks: 22 (limit: 2062)
   CGroup: /system.slice/google-assistant.service
           └─479 /home/pi/env/bin/python -u /home/pi/Assistants-Pi/Google-Assist

Jul 11 19:00:23 raspberrypi python[479]: jack server is not running or cannot be
Jul 11 19:00:23 raspberrypi python[479]: JackShmReadWritePtr::~JackShmReadWriteP
Jul 11 19:00:23 raspberrypi python[479]: /home/pi/env/lib/python3.7/site-package
Jul 11 19:00:23 raspberrypi python[479]:   warnings.warn('Google Assistant Libra
Jul 11 19:00:25 raspberrypi python[479]: device_model_id: assistant-5d0d6-prasan
Jul 11 19:00:25 raspberrypi python[479]: device_id: D746B566ADC28701F865678C10F6
Jul 11 19:00:27 raspberrypi python[479]: ON_MUTED_CHANGED:
Jul 11 19:00:27 raspberrypi python[479]:   {"is_muted": false}
Jul 11 19:00:29 raspberrypi python[479]: ON_START_FINISHED
Jul 11 19:00:29 raspberrypi python[479]: ON_MEDIA_STATE_IDLE
 
And for Pulse audio,

>pi@raspberrypi:~ $ sudo systemctl status pulseaudio.service
● pulseaudio.service - PulseAudio system server
   Loaded: loaded (/etc/systemd/system/pulseaudio.service; enabled; vendor prese
   Active: active (running) since Sun 2021-07-11 19:00:12 IST; 6min ago
 Main PID: 371 (pulseaudio)
    Tasks: 4 (limit: 2062)
   CGroup: /system.slice/pulseaudio.service
           └─371 /usr/bin/pulseaudio --daemonize=no --system --realtime --log-ta

Jul 11 19:00:08 raspberrypi pulseaudio[371]: Running in system mode, but --disal
Jul 11 19:00:08 raspberrypi pulseaudio[371]: Running in system mode, forcibly di
Jul 11 19:00:08 raspberrypi pulseaudio[371]: Running in system mode, forcibly di
Jul 11 19:00:08 raspberrypi pulseaudio[371]: OK, so you are running PA in system
Jul 11 19:00:08 raspberrypi pulseaudio[371]: Please read http://www.freedesktop.
Jul 11 19:00:11 raspberrypi pulseaudio[371]: Failed to open cookie file '/var/ru
Jul 11 19:00:11 raspberrypi pulseaudio[371]: Failed to load authentication key '
Jul 11 19:00:11 raspberrypi pulseaudio[371]: Failed to open cookie file '/var/ru
Jul 11 19:00:11 raspberrypi pulseaudio[371]: Failed to load authentication key '
Jul 11 19:00:12 raspberrypi systemd[1]: Started PulseAudio system server.

All of them shows active but I did not not hear the Hi sound and music from any of them and also not responding for wakeword.
Alexa is not responding or say hi even I start manually.
But google assistant works well when I start manually but one problem is its voice is low sound.

shivasiddharth, 2021-07-11:
Did you authenticate Alexa ? Start alexa manually and share the screenshot.
First time when you start Google Assistant, you need to give the command to set volume to maximum.

Prasanna-16, 2021-07-11:
Here is the screen shot sir.
[image]

Prasanna-16, 2021-07-11:
Also now the google assistant is also not responding.
I shared the screen shot sir.
It says on the code cannot to server jack.
please help me out.
[image]

Prasanna-16, 2021-07-11:
> Did you authenticate Alexa ? Start alexa manually and share the screenshot.
> First time when you start Google Assistant, you need to give the command to set volume to maximum.

I authenticated alexa already sir.

shivasiddharth, 2021-07-11:
I need to see the last part of Alexa screen. First half does not help.

Prasanna-16, 2021-07-11:
Ok sir. Here is it.
[image]
In this image the last line is repeating again and again.

shivasiddharth, 2021-07-11:
Its not detecting your mic. Post output of:
```
aplay -l
arecord -l
```

Prasanna-16, 2021-07-12:
It shows like this sir.
[image]

shivasiddharth, 2021-07-12:
Share the output of:
```
sudo nano ./.asoundrc
sudo nano /etc/asound.conf
```

Prasanna-16, 2021-07-12:
Sir it shows like this for sudo nano ./.asoundrc

[image]

And for sudo nano /etc/asound.conf like this
[image]

shivasiddharth, 2021-07-12:
Its an issue with your Mic. Try changing the Mic.

Prasanna-16, 2021-07-12:
Ok sir but assistants not starting on boot.
Help with this issue.

shivasiddharth, 2021-07-12:
> Ok sir but assistants not starting on boot.
> Help with this issue.

Assistants are indeed starting. The "active" that you see in the status means that the assistants are starting on boot. They are not able to utilize the Mic. So there is some issue with the Mic (Issue is the MIC). It could be a software or hardware issue. To see if it is a hardware issue I asked you to change the Mic. If the new Mic also does not work, then its a software issue.
I dont understand as to what more you want me to do.

Prasanna-16, 2021-07-12:
Ok sir thankyou. I am going to try configure the mic.

shivasiddharth, 2021-07-13:
If you have Pulseaudio as systemwide there will be no conflict. Manually start the Alexa and post the output of the last part.

Prasanna-16, 2021-07-13:
Thankyou sir. After I had set Pulse audio as system wide, It both google and alexa responds for my command. But Alexa voice is not good. It's voice sounds like a radio static. Like broken voice. What is causing this issue sir?

Prasanna-16, 2021-07-13:
Ok sir.

Prasanna-16, 2021-07-30:
Finally, After a whole complete reinstall, everything works fine.
Thank you sir.

raspf450, 2021-08-09:
> Finally, After a whole complete reinstall, everything works fine.
> Thank you sir.

Hey, can u help me with the same. Which version of rasbian os are you using? And how did you solve the broken voice issue?

Prasanna-16, 2021-08-10:
Just reinstall with a new config.json file(create another product).

raspf450, 2021-08-10:
> Just reinstall with a new config.json file(create another product).

Just reinstall alexa and assistant or I need to reinstall the rasbian as well?

Prasanna-16, 2021-08-10:
Reinstall rasbian also to delete old assistant files fully.

Prasanna-16, 2021-08-12:
Till now alexa started on boot perfectly. But now, it is showing like this
[image]
[image]
 Please help sir