shivasiddharth/Assistants-Pi #170: cant perform step 10

sushant96702, 2021-07-17:
while executing command 
sudo /home/pi/Assistants-Pi/Alexa/startsample.sh  
it gave me error 
sudo: /home/pi/Assistants-Pi/Alexa/startsample.sh: command not found
startsample.sh is not there in alexa folder

shivasiddharth, 2021-07-17:
The assistant has not been compiled properly. Re-run the installer.

closed 2021-07-17 by shivasiddharth

sushant96702, 2021-07-17:
> The assistant has not been compiled properly. Re-run the installer.

i reran but still same error

shivasiddharth, 2021-07-17:
Manually open the Alexa folder and see if startsample.sh file is there or not. You need to watch the terminal. There might have been some compilation error. Just Re-running the installer will not fix any compilation issue, its not some magic step/command.

sushant96702, 2021-07-19:
bro i checked all files it was not there : /

shivasiddharth, 2021-07-19:
So its a compilation issue. You need to run the installer and watch for errors. If its not compiling properly it will for sure point out the error.

kashifabidi, 2022-05-29:
> So its a compilation issue. You need to run the installer and watch for errors. If its not compiling properly it will for sure point out the error.

These are the only errors faced through install

chmod: cannot access '/home/root/Assistants-Pi/audio-drivers/USB-MIC-HDMI/scripts/configure.sh': No such file or directory
sudo: /home/root/Assistants-Pi/audio-drivers/USB-MIC-HDMI/scripts/configure.sh: command not found
chmod: cannot access '/home/root/Assistants-Pi/audio-drivers/USB-MIC-HDMI/scripts/install-usb-mic-hdmi.sh': No such file or directory
sudo: /home/root/Assistants-Pi/audio-drivers/USB-MIC-HDMI/scripts/install-usb-mic-hdmi.sh: command not found
Audio set to be forced through HDMI.

kashifabidi, 2022-05-29:
Looking at OP's repository, the file isn't even on GIT

kashifabidi, 2022-05-29:
> while executing command sudo /home/pi/Assistants-Pi/Alexa/startsample.sh it gave me error sudo: /home/pi/Assistants-Pi/Alexa/startsample.sh: command not found startsample.sh is not there in alexa folder

https://github.com/alexa/avs-device-sdk/issues/1003#issuecomment-536204106

Found this link somewhere, I am trying it now. Hope it helps us both.