shivasiddharth/Assistants-Pi #172: RPI4 AUX output not listed after prep.sh

iamjmat, 2021-10-30:
During prep.sh I gave AIY-HAT as the audio and mic configuration and after reboot, I am not able to get my AUX output.

Below is my headset and speaker configuration that I intend to use with my RPI4:
1. Mic input through the USB headset (JABRA Evolve 30 II) that I have connected
2. Speaker output through the speaker that I connected via AUX

Here is my arecord and aplay list command outputs 
```
pi@jarvis:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sndrpigooglevoi [snd_rpi_googlevoicehat_soundcar], device 0: Google voiceHAT SoundCard HiFi voicehat-codec-0 [Google voiceHAT SoundCard HiFi voicehat-codec-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: II [Jabra EVOLVE 30 II], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@jarvis:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpigooglevoi [snd_rpi_googlevoicehat_soundcar], device 0: Google voiceHAT SoundCard HiFi voicehat-codec-0 [Google voiceHAT SoundCard HiFi voicehat-codec-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: II [Jabra EVOLVE 30 II], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
How to revert back so that AUX output works and which configuration option should I choose for my configuration?

iamjmat, 2021-10-30:
Seems like prep.sh removed some lines in my boot/config.txt when I gave the AIY-HAT option. After reverting those changes , I am able to get the AUX output listed. However, I still need help regarding the configuration option which would be suitable for me

Also, suggestion to make those lines commented rather than removing in prep.sh.

shivasiddharth, 2021-10-30:
Certain lines have to be commented in the config file for the AIY HAT to work. The script has done its work correctly.   It's not an Auxillary port on the board. It's a Jack. You had to select USB Mic and Jack configuration.

closed 2021-10-30 by shivasiddharth